deepguider/DeepGuider #21: output of vps module

roricljy, 2020-02-24:
Currently, the VPS module gives the best matched image and its matching error(difference measure with the input image) as its output. However, it should be revised to output 'confidence' of the matching in the rage of [0, 1] instead of 'error' metric for global consistency.

ccsmm78, 2020-03-10:
Currently, confidence values mean 1/distance. I'm going to replace it with a new function instead of just 1/distance.

ccsmm78, 2020-03-10:
VPS uses and returns similarity(confidence) for searching correspondence, instead of L2 distance, now.
Thing changed:
faiss.IndexFlatIP() instead of faiss.IndexFlatL2

Refer to : https://github.com/facebookresearch/faiss/wiki/Faiss-indexes

closed 2020-03-10 by ccsmm78