deepguider/DeepGuider #34: Revision of data structure and representation of dg::Path and dg::Map

roricljy, 2020-03-05:
Current:
 - DeepGuider system
  . dg::**Map** consists of **Node**s and **Edge**s
  . Each **Node** has unique ID but **Edge** is identified by the adjacent two **Node**s
  . **Path** is defined as an ordered list of **Node** ID's
 - Map Server
  . Assign unique ID to Edge as well as Node
  . Path is generated to include both Nodes and Edges

Problem:
 - In DeepGuider system, accessing the Nodes and Edges from the Path is complex and inconvenient
   . first, searching two adjacent Nodes on the Path from the **Map** by their Node ID's
   . then, identifying common Edge by matching the edge lists of the searched two Nodes.
 - Map and Path representation is inconsistent between DeepGuider system and Map Server

Possible revision:
 - open problem to discuss

zeozeo7142, 2020-03-05:
효율적인 측면에서 dg::Path내에서 Node의 ID만 아니라 다른 정보들도 가지고 있는 것이 좋지 않을까 싶습니다. 어차피 Routing Server에서 Path상의 Node들에 대한 정보를 통째로 주고 있는 상황에서 일부러 정보를 받지 않고 Node ID로 dg::Map에서 다시 정보를 Node마다 일일이 검색하는 것은 좀 비효율적인 것 같습니다. 하지만 Routing Server에서 주는 Path상의 Edge 정보를 가지고 있거나 직접 이용하는 것은 원래 Node의 하위 개념인 Edge의 의미에 영향을 줄 수도 있다는 점에서 좀더 신중해야 될 것 같습니다.

seohyunatwork, 2020-03-05:
Guidance module needs node and edge information to generate guide, such as edge type and  distance. It will be convenient if the guidance module can directly access node and edge contents.(e.g. findEdge(ID edgeid))

sunglok, 2020-03-05:
Anyway, it is (potentially) helpful to contain and use more information (e.g. edge ID) if it is given. Let's take an example such as the portal map service by NAVER and Google. Moreover, what about map definition in OpenStreetMap, IEEE MDR, and ROS? I will update this post if I find them.

roricljy, 2020-03-06:
I also agree with @zeozeo7142 and @seohyunatwork.
How about modify the code like the below to fulfill everyone's requirements?

```
struct PathNode {
    Node* node;
    Edge* edge; // edge==nullptr for last path node
};
class Path {
    vector<PathNode> pnode;
};
```

roricljy, 2020-03-06:
My previous suggestion is minimal modification to keep the current data structures. However, if we have time and want to whole revision, I think the following revision also can be a solution:

```
class Node {
    ID id;
    vector<Edge *> edge_list; // include both of in_edges & out_edges
    ...
};
class Edge {
    ID id;
    Node* from;
    Node* to;
    bool directed; // directed==true if it is one-way edge
    ...
};
class Map {
    list<Node> nodes;
    list<Edge> edges;
};
struct PathNode {
    Node* node;
    Edge* edge; // edge==nullptr for last path node
};
class Path {
    vector<PathNode> pnode;
};
```

zeozeo7142, 2020-03-06:
I think, even if it takes time it is better to modify the current data structure as a whole.

sunglok, 2020-03-10:
I have completed to apply the above interface at ccd4a0328843c8346a47d444ef96da5d1f192f73

closed 2020-03-10 by sunglok