deepguider/DeepGuider #38: Debugging Message Off [Guidance & VPS]

roricljy, 2020-03-10:
Currently, VPS module and Guidance module prints out lots of debugging messages at runtime.
The debugging messages of each module can be helpful to unit test but it makes hard to debug the integrated system.
It would be better to turn off the message print by default or to provide a function to turn it on and off.

ccsmm78, 2020-03-10:
I removed all the vps intermediate result output. 
If you want to see intermediate results, you can set the --verbose option to True in vps.py.

seohyunatwork, 2020-03-12:
printf for debugging in guidance.cpp is removed.
There is a printf in "printSingleRGuide"method, which is to show the final guidance, not for debugging.

roricljy, 2020-03-23:
POI module also need to provide functionality of turning its debugging messages on and off programmatically.

roricljy, 2021-09-30:
resolved

closed 2021-09-30 by roricljy