deepguider/DeepGuider #41: Suppress warning messages

roricljy, 2020-03-10:
The follows codes cause warning messages at compile time:
- (graph_painter.hpp:322)  cv::putText(image, cv::format("%d", n->data.id), p + font_offset, cv::FONT_HERSHEY_DUPLEX, font_scale, font_color, int(font_scale));
  . %d --> %zu ?
- (guidance.cpp:77) printf("\curedgeid=%llu\n", curedgeid);
  . "\curedgeid --> "curedgeid ?
  . %llu --> %zu
- (guidance.cpp:100): if (afterNextNodeId == NULL)	//last node check
  . type of afterNextNodeId is dg::ID and dg::ID == NULL doesn't make sense.
  . anyway to remove warning, 'nullptr' can be used in c++ instead of 'NULL'

sunglok, 2020-03-11:
I didn't get any warning with `cv::putText` with `%d` and `dg::ID`.

roricljy, 2020-03-11:
@sunglok With `%d`, it gives the following warning messages in Linux build:
```
/home/dg/work/deepguider/examples/dg_test/../../src/core/graph_painter.hpp: In instantiation of ‘static bool dg::GraphPainter<D, C>::drawNodes(cv::Mat&, const dg::CanvasInfo&, dg::GraphPainter<D, C>::Point2IDGraph&, double, double, const Vec3b&, int) [with D = dg::Point2ID; C = double; dg::GraphPainter<D, C>::Point2IDGraph = dg::DirectedGraph<dg::Point2ID, double>; cv::Vec3b = cv::Vec<unsigned char, 3>]’:
/home/dg/work/deepguider/examples/dg_test/../../src/core/graph_painter.hpp:163:22:   required from ‘bool dg::GraphPainter<D, C>::drawMap(cv::Mat&, dg::GraphPainter<D, C>::Point2IDGraph&) const [with D = dg::Point2ID; C = double; dg::GraphPainter<D, C>::Point2IDGraph = dg::DirectedGraph<dg::Point2ID, double>]’
/home/dg/work/deepguider/examples/dg_test/main.cpp:188:5:   required from here
/home/dg/work/deepguider/examples/dg_test/../../src/core/graph_painter.hpp:322:46: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘dg::ID {aka long unsigned int}’ [-Wformat=]
                 cv::putText(image, cv::format("%d", n->data.id), p + font_offset, cv::FONT_HERSHEY_DUPLEX, font_scale, font_color, int(font_scale));
```
I'm not sure of it is specific to compiler vision..

seohyunatwork, 2020-03-12:
(guidance.cpp:77) printf("\curedgeid=%llu\n", curedgeid); ==> removed
(guidance.cpp:100): if (afterNextNodeId == NULL) //last node check ==> removed

sunglok, 2020-03-24:
I did not get the warning (MSVC 2019 and Linux with OpenCV v3.2), but I fixed it with other warnings. Please refer fcf3c28d67f673f7908c77da3eb2b9a4d8da274d.

closed 2020-03-24 by sunglok