deepguider/DeepGuider #42: Runtime error (VPS)

roricljy, 2020-03-10:
When the VPS is called repeatedly, it fails after several successful initial run with the following error message:
```
iteration: 0 (it took 1.067000 seconds)
        top0: id=2813410193600000, confidence=0.62, t=1583826315.692000
        top1: id=2813410193300000, confidence=0.62, t=1583826315.692000
        top2: id=2813410191700000, confidence=0.61, t=1583826315.692000
iteration: 1 (it took 1.041000 seconds)
        top0: id=2813410193600000, confidence=0.62, t=1583826316.759000
        top1: id=2813410193300000, confidence=0.62, t=1583826316.759000
        top2: id=2813410191700000, confidence=0.61, t=1583826316.759000
iteration: 2 (it took 1.026000 seconds)
        top0: id=2813410193600000, confidence=0.62, t=1583826317.800000
        top1: id=2813410193300000, confidence=0.62, t=1583826317.800000
        top2: id=2813410191700000, confidence=0.61, t=1583826317.800000
iteration: 3 (it took 1.029000 seconds)
        top0: id=2813410193600000, confidence=0.62, t=1583826318.826000
        top1: id=2813410193300000, confidence=0.62, t=1583826318.826000
        top2: id=2813410191700000, confidence=0.61, t=1583826318.826000
ERROR: Unexpected segmentation fault encountered in worker.
Traceback (most recent call last):
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 761, in _try_get_data
    data = self._data_queue.get(timeout=timeout)
  File "/usr/lib/python3.6/queue.py", line 173, in get
    self.not_empty.wait(remaining)
  File "/usr/lib/python3.6/threading.py", line 299, in wait
    gotit = waiter.acquire(True, timeout)
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/_utils/signal_handling.py", line 66, in handler
    _error_if_any_worker_fails()
RuntimeError: DataLoader worker (pid 6632) is killed by signal: Segmentation fault. 

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./../src/vps/vps.py", line 631, in apply
    recalls = self.test_dg(whole_db_set,whole_q_set, epoch, write_tboard=False)
  File "./../src/vps/vps.py", line 340, in test_dg
    dbFeat = self.test_sub(eval_set_db,epoch=epoch)
  File "./../src/vps/vps.py", line 301, in test_sub
    for iteration, (input, indices) in enumerate(test_data_loader, 1):
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 841, in _next_data
    idx, data = self._get_data()
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 798, in _get_data
    success, data = self._try_get_data()
  File "/work/venv/dg/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 774, in _try_get_data
    raise RuntimeError('DataLoader worker (pid(s) {}) exited unexpectedly'.format(pids_str))
RuntimeError: DataLoader worker (pid(s) 6632) exited unexpectedly

If you suspect this is an IPython 7.11.1 bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@python.org

You can print a more detailed traceback right now with "%tb", or use "%debug"
to interactively debug it.

Extra-detailed tracebacks for bug-reporting purposes can be enabled via:
    %config Application.verbose_crash=True

VPS::apply() - Call failed
The terminal process terminated with exit code: 255
```

roricljy, 2020-03-10:
I'm not sure if it is a problem related to running environment (system) or VPS module itself.

ccsmm78, 2020-03-10:
I get a similar error in my environment.
However, it doesn't happen when vps.py itself is executed in a loop (10 times). I'll debug this thing.

ccsmm78, 2020-03-11:
Fixed.
Refer to :
https://github.com/pytorch/pytorch/issues/4969

closed 2020-03-11 by ccsmm78