deepguider/DeepGuider #46: Node and Edge type in dg namespace

seohyunatwork, 2020-03-11:
Types of Node and Edge are accessible directly from dg namespace in "map.hpp". (ex. dg::NT_BS)
Is it OK this way or how about levelling down to dg::NodeType::NT_BS? 
@sunglok

roricljy, 2020-03-12:
Good issue. Can you explain more about why this change is necessary?

seohyunatwork, 2020-03-16:
My opinion is, it will be more readable if similar properties are collected in one category.
For now, when typing 'dg::', it shows lots of child contents. 
If a user only wants to see the type of node, 'dg::NodeType::' might help. 
But, I am also not sure whether making this into one category is better way. (dg::NodeType::NT_BS)
Maybe there is a reason to leave it currently (ex. dg::NT_BS)
Also, the name 'dg::Nodetype is' already reserved in 'directed_graph.hpp'.
So, I am not sure.

sunglok, 2020-03-17:
In addition, how about using **more longer** but **more clear** names? (e.g. `NT_BS` to `NODE_BASIC` or `NODE_TYPE_BASIC`) Readability is the reason why we use enumeration. Now they are too short so hard to understand by watching them first time.

roricljy, 2020-03-18:
Thank @seohyunatwork for the explanation. I agree with all the idea and I was also worried about the same problem at the initial implementation. One possible way of modification would be to embed node types to **dg::Node::** and edge types to **dg::Edge::**, respectively.

I also agree with using more loner and concrete names. I reviewed it again and suggest the following revision:
// node types
NT_BS --> NODE_BASIC
NT_JT --> NODE_JUNCTION (Nodes where two or more roads/paths meet)
(      )  --> NODE_CORNER (new node type by separating it from NT_JT)
NT_DR --> NODE_DOOR
NT_EV --> NODE_ELEVATOR
NT_ES --> NODE_ESCALATOR

// edge types
ET_SD --> EDGE_SIDEWALK
ET_MD --> EDGE_MIDDLEWAY (indoor: corridor, outdoor: single road without boundary between the vehicle road and pedestrian road)
ET_CR --> EDGE_CROSSWALK
ET_DR --> (remove doorway. it can be substituted by EDGE_MIDDLEWAY)
ET_EV --> EDGE_ELEVATOR
ET_ES --> EDGE_ESCALATOR
(      )  --> EDGE_STAIR (new edge type)

roricljy, 2020-03-19:
The revision was made and applied as follows:

1. Node type enumeration is embedded to **dg::Node::** and edge types to **dg::Edge::**, respectively.

2. full-naming of node types and edge types
// node types
NT_BS --> NODE_BASIC
NT_JT --> NODE_JUNCTION
NT_DR --> NODE_DOOR
NT_EV --> NODE_ELEVATOR
NT_ES --> NODE_ESCALATOR

// edge types
ET_SD --> EDGE_SIDEWALK
ET_MD --> EDGE_ROAD (indoor: corridor, outdoor: single road without boundary between the vehicle road and pedestrian road)
ET_CR --> EDGE_CROSSWALK
ET_DR --> (removed)
ET_EV --> EDGE_ELEVATOR
ET_ES --> EDGE_ESCALATOR
( ) --> EDGE_STAIR (added as new edge type)

closed 2020-03-19 by roricljy