deepguider/DeepGuider #47: Pointer to temporal instance (MapManager::generatePath)

roricljy, 2020-03-16:
In function MapManager::generatePath(...), PathElement::node and PathElement::edge are initialized with the address of temporal local variables which are not valid after termination of the function.

They should be initialized to point out the Nodes and Edges in the downloaded Map.

seohyunatwork, 2020-03-16:
First, we decided to include only Node IDs in the path.
Second, we decided to include both Node's and Edge's IDs.
Now that, there is PathElement with Node* and Edge*, 
I think its OK to change the path with PathElement.

sunglok, 2020-03-17:
Now, `dg::Path` contains points to nodes and edges. It is dangerous and ambiguous whether `dg::Path` indicates valid instances or not. How about using IDs instead of pointers.

zeozeo7142, 2020-03-25:
I solved this issue #47.

closed 2020-03-25 by zeozeo7142