deepguider/DeepGuider #49: [VPS]  If apply () is called repeatedly using the actual query image, segmentation fault occurs.

ccsmm78, 2020-03-17:
If apply () is called repeatedly using the actual query image, segmentation fault occurs.
When tested with only python code, there is no problem, but it is a problem that occurs in code (example) that works with C ++.

- (close) The first issue, a broken image is often included in the query image, so logic to check it is added to vps.py.

-  (open) The time when an error occurs usually occurs after 10 iterations. However, it often occurs after 20 or 15 times, so it is not consistent.

**_The reason for including other assignees is that this issue is still being analyzed, so be careful to use it._**

roricljy, 2020-03-17:
Yes.. segmentation fault still occurs.
* _vps_test_ stops after 2 iterations with segmentation fault. (with sample query image & image server)
* _dg_test_ stops after 17 iterations with segmentation fault. (with ETRI dataset & image server)

I tested it several times and always have the same result. Also, I changed query images in _dg_test_ but it stop exactly at 18th iteration.

ccsmm78, 2020-03-17:
So far, it seems to be a segmentation fault that occurs when connecting to the image server.
Except for the code that connects to the image server, the problem does not occur even over 300 iterations for a half hour.
requests.get(req_str,timeout=timeout) in get_streetview.py

**To : Jaeho Lim.**
The map server needs to be tested for a stability. An experiment to download streetview images continuously for an hour.

ccsmm78, 2020-03-18:
An experiment in progress that severely calls request to the naver image server in get_streetview.py

ccsmm78, 2020-03-19:
Issue #49 was nearly fixed: Segmentation fault occurs irregularly in vps_test.

When C and python are integrated, there seems to be a bug between some
packages. For example, scipy's loadmat / savemat caused a segmentation
fault in python + 'C ++' environment.

Fixed 1 : Instead of saving the db image list as a mat file, modify them into
 dictionary variables.

To do : In addition, in the case of requests.get(), when it is called within
 complex python code, it causes a segmentation fault, but it has not yet
 been confirmed which package is in conflict with.

ccsmm78, 2020-03-23:
In an environment where Python is embedded in C, it is thought that some
IO-related packages in Python cause segmentation faults.
To solve this, some packages were removed or replaced by os.system ()
call. After this modification, it was confirmed that there were no
errors in about 4000 consecutive query inputs.
 - remove : scipy.savemat, scipy.loadmat, glob.glob()
 - replace : requests.get(), json ==> os.system("curl URL") or subprocess.check_out("curl URL)
             requests.get(), dump ==> os.system("curl URL --output SVID.jpg")
No reports and any resolutions about this issue were found on the web. Therefore, the method of solving the problem may not be official. However, it was solved by the method described above.

closed 2020-03-23 by ccsmm78