deepguider/DeepGuider #50: [cx::Algorithm] setParamValue() can cause error depending on OpenCV version

roricljy, 2020-03-19:
In Linux and OpenCV 3.3, cx::Algorithm::setParamValue(const std::string& __param__, double value) cause error when the type of member variable associated with the __param__ name is _int_.

example case:
```
int GraphPainter::m_edge_thickness;
GraphPainter::setParamValue("edge_thickness", 2);
// internally, it generates yml string -> "edge_thickness: 2.00000"
// and then parsed into int type variable (not double type).
// after the execution of setParamValue, m_edge_thickness became very very large value unexpectedly. 
```
I'm not sure what will happen with other opencv versions or on windows.

sunglok, 2020-03-20:
I expect that `double` to `int` might be no problem. I will check and update its progress.

roricljy, 2020-04-02:
I don't know why but it occurs when the code is executed in ROS.

roricljy, 2021-09-30:
resolved (it works in ros melodic)

closed 2021-09-30 by roricljy