deepguider/DeepGuider #51: [Examples] add README.md

roricljy, 2020-03-20:
We need to add README.md for each project in deepguider/examples, which includes
- setup instructions and
- how to build and run.

roricljy, 2020-03-20:
[dg_test] it needs to install curl
  -> $sudo apt install libcurl

ccsmm78, 2020-03-20:
example of build_and_run.sh 

#!/bin/bash
##Current Directory : topdir/examples/vps_test
cd ../../
topdir=`pwd`

echo "####### Compile ..... #######"
cd ${topdir}/examples/vps_test
mkdir -p build
cd build
cmake ..
make install
if [ $? -ne 0 ];then # If it met error in previous step
    #echo $?
    exit 0
fi

echo "####### Run ..... #######"
cd "$topdir/bin"
ln -sf ../src/vps/data_vps
source ~/.virtualenvs/dg_venv3.6/bin/activate
#gdb ./vps_test #for debugging, you need to insert '-g' option in your CMakeLists.txt : set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -g")
#kdbg ./vps_test #for debugging in graphic mode
./vps_test # for  running without debugging

roricljy, 2020-03-24:
@ccsmm78  great example!

ccsmm78, 2020-03-25:
I added build_all.sh which builds all unit modules in sub-directories at DeepGuider/examples.

DeepGuider/examples/build_all.sh

#!/bin/bash

function build_unit(){
    D=$1
    cd ./$D
    mkdir -p build
    cd build
    cmake ..
    make install
    cd ../../
}

function build_script(){
    D=$1
    found_script=`ls *build*.sh 2> /dev/null|grep -v 'run'`
    if [ ! -z "$found_script" ];then
        ./$found_script
    fi  
    cd ..
}

counter=0
for D in $(find . -mindepth 1 -maxdepth 1 -type d)
do
    counter=`expr $counter + 1`;
    echo "[$counter]################# Enter and Compiling [$D] ####################"
    echo ""
    build_unit "$D"
done

roricljy, 2020-03-26:
Awesome. I learn a lot ~~
Initially, I used _topdir=pwd_ by referring to your code but it doesn't work in my environment. _topdir_ set to be string "pwd" instead of output of _pwd_. So I simply had to use relative paths..

roricljy, 2020-04-28:
done

closed 2020-04-28 by roricljy

seohyunatwork, 2020-05-15:
Done in Guidance module.