deepguider/DeepGuider #52: [POI] segmentation fault

roricljy, 2020-03-23:
When the POI module invoked with real image, it sometimes causes segmentation fault.

Success Case I. Test-Run with sample image:
```
Found 7 boxes for img
logo 0.06 (50, 192) (94, 236)
logo 0.08 (1027, 751) (1105, 768)
logo 0.26 (370, 392) (965, 593)
logo 0.30 (1049, 718) (1088, 747)
logo 0.83 (397, 585) (437, 645)
logo 0.95 (47, 0) (239, 197)
logo 0.95 (663, 286) (839, 441)
Time spent: 0.322sec
Found 7 logos from 7 classes
Logo #6 - (663, 286) (839, 441) - classified as costa 1.00
Logo #4 - (397, 585) (437, 645) - classified as burgerking-symbol 1.00
Logo #5 - (47, 0) (239, 197) - classified as burgerking-symbol 1.00
```

Success Case II. Test-Run with real image:
```
Found 2 boxes for img
logo 0.19 (754, 141) (768, 159)
logo 0.20 (278, 345) (304, 361)
Time spent: 0.267sec
Found 2 logos from 2 classes

```

Failure Case. Test-Run with another real image:
```
Found 1 boxes for img
logo 0.38 (111, 338) (162, 355)
Time spent: 0.250sec
Found 1 logos from 1 classes
/bin/bash: line 1:  7926 Segmentation fault      (core dumped) ./dg_test
The terminal process terminated with exit code: 139
```

It seems that the POI gives error when the detected logo with high confidence (>0.3 ?) failed to be classfied (maybe not learned? or haven't registered?)

minnjung, 2020-03-23:
I don't know why the semengation fault is going to happen, but it doesn't seem to be related to confidence score. When I tried it, this problem didn't happen. I'll have to check a little bit more about this problem, but it would be helpful if you could tell me more about the test environment.

roricljy, 2020-03-24:
It might be a problem of environment where Python is embedded in C++. I updated _examples/poi_recog_test_ to support video input. Currently it causes segmentation fault even with a simple image input when it is called iterative. We also had similar problem with VPS module, which is now solved partially. Please check issue #49 .

roricljy, 2020-03-26:
I found some symbolic links are missing in the shell scrip _build_run_poi.sh_ in **examples/poi_recog_test/**. I uploaded the updated script. You can test your python code in c++ embedding environment by simply run _exmaples/poi_recog_test/build_run_poi.sh_.

As you can see it in the below, we need to create lots of symbolic links to run poi module in deepguider system. To make the problem simple, I ask you to use single data folder, named _data_poi_ in the next version of POI.
```
#!/bin/bash
##Current Directory : topdir/examples/poi_recog_test
echo "####### Compile ..... #######"
mkdir -p build
cd build
cmake ..
make install
if [ $? -ne 0 ];then # If it met error in previous step
    #echo $?
    cd ..
    exit 0
fi

echo "####### Run ..... #######"
cd ../../../bin
ln -sf ../src/poi_recog/model
cd data
ln -sf ../../src/poi_recog/data/test
ln -sf ../../src/poi_recog/data/preprocessed
ln -sf ../../src/poi_recog/data/litw_annotation.py
cd ..

#gdb ./poi_recog_test #for debugging, you need to insert '-g' option at CXX_FLAGS in your CMakeLists.txt # in your CMakeLists.txt : set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -g")
#kdbg ./poi_recog_test #for debugging in graphic mode
./poi_recog_test # for running without debugging

cd ../examples/poi_recog_test   # return to initial directory
```

roricljy, 2020-04-22:
It was the problem of c++ python wrapper.
I pushed corrected _poi_recognizer.hpp_ and it works well now.

closed 2020-04-22 by roricljy