deepguider/DeepGuider #56: [MapManager] server ip setting

roricljy, 2020-03-25:
1. Bug fix: in the function MapManager::generatePath(), **getMap(path)** should be **getMap(path, ip)**
```
	ok = parsePath(json);
	if (!ok) return false;

	Path path = m_path;
	getMap(path); ----> getMap(path, ip);
```

2. Revision suggestion
I think it's also good to use local variable __ip_ and share it across the functions instead of providing it as argument of each function... In that case, the default server ip can be changed explicitly by some api call when it is necessary.

roricljy, 2020-03-25:
It took 2 hours to find this.. t.t At the beginning, I thought that my curl library or network has problem..

zeozeo7142, 2020-03-25:
I'm really sorry! I thought I was looking at my codes as a whole, but I couldn't find it(getMap(path)).

zeozeo7142, 2020-03-25:
I made setter and getter functions for server IP address.

closed 2020-03-25 by zeozeo7142