deepguider/DeepGuider #57: [VPS] configurable ip address

roricljy, 2020-03-25:
Currently, the ip address of streetview server has to be modified manually in the code, _vps.py_.
It will be better if we can configure the ip address programmatically. 
It is undesirable that the modified _vps.py_ is submitted together with other module's code when someone update his/her own code and then submits it.

ccsmm78, 2020-03-25:
Yes, you heat the good points.
I've added follow things to do it:

You can set "image server's ip address" programmatically in example/vps_test/main.cpp :
   25     //const char* ipaddr = "129.254.87.96";
  26     const char* ipaddr = "localhost";

This is used vps.apply() to set server's ip address :
  71         if (!vps.apply(image, N, gps_lat, gps_lon, gps_accuracy, t, **_ipaddr_**)) {
  72             return -1;
  73         }

In, detail, I used PyUnicode_FromString(ipaddr) in example/vps_test/vps.hpp
  77             // Image server's ip address
  78             pValue = **_PyUnicode_FromString_**(ipaddr);
  79             PyTuple_SetItem(pArgs, arg_idx++, pValue);

closed 2020-03-25 by ccsmm78

roricljy, 2020-03-26:
Good job!! and thanks for modifying vps.hpp together. It is very helpful.