deepguider/DeepGuider #58: [VPS_TEST] no response in thread run

roricljy, 2020-03-25:
The vps module doesn't respond when it is run as a thread.
The thread run is implemented and can be tested at examples/vps_test/main.cpp

roricljy, 2020-03-25:
I am sorry to raise issue related to running environment.. but it has to be resolved to run deepguider system in real mode.

ccsmm78, 2020-03-25:
Awesome code!!,  vps_test/main.cpp.
In particular, it is able to test various conditions. I love it.
    bool test_image = false;
    bool test_video = false;
    bool test_query = true;
    bool test_thread_run = false;
    bool enable_recording = false;

Unfortunately, considering the capacity and management of the DB, the streetview DB is stored in ramdisk on-demand.
So, I'm coding assuming that only one vps thread is running. If more than one vps needs to be threaded, please let me know in the comments.

roricljy, 2020-03-26:
No problem. For the moment, it is sufficient to test it in single thread.

ccsmm78, 2020-03-26:
It may need some thread related API to run "embedded python", such as  PyEval_InitThreads(), PyEval_SaveThread() and so on.

I added some wrapper "thread_apply()" for apply() :
        /**
         * for thread of CPP
         */
        bool thread_apply(cv::Mat image, int N, double gps_lat, double gps_lon, double gps_accuracy, dg::Timestamp t, const char* ipaddr)
        {
            PyGILState_STATE state;
            bool ret;

            if (!m_gil_init){
                m_gil_init = 1;
                PyEval_InitThreads();
                PyEval_SaveThread();
            }
            state = PyGILState_Ensure();

            /* Call Python/C API functions here */
            ret = apply(image, N, gps_lat, gps_lon, gps_accuracy, t, ipaddr);

            PyGILState_Release(state);

            return ret;
        }

Then  I call vps.thread_apply() instead of vps.apply() in while loop in main.cpp as follows :
113 :  VVS_CHECK_TRUE(vps.thread_apply(image, N, gps_lat, gps_lon, gps_accuracy, t1, map_server_ip.c_str())); instead of vps.apply()

Then it works.

**If there are no problems after review, please close this issue.**

ccsmm78, 2020-03-26:
Success in an aging test over 2000 frames of 191115_ETRI.avi

roricljy, 2020-03-26:
You rock!!
It works well now. I will close this issue after more revision of code and reviewing other dependencies.

closed 2020-03-27 by ccsmm78

roricljy, 2020-04-01:
I've revised the code to initialize the python environment with optional parameter of _support_thread_run_ and thread handling works implicitly.