deepguider/DeepGuider #59: get StreeView Image

roricljy, 2020-03-25:
It will be very useful for GUI display and debugging if there is an api to get streetview image by dg::ID in a form of cv::Mat.

ccsmm78, 2020-03-25:
There is a getStreetView (lat, lon, radius) function in map_manager, whose return value is m_map-> views.
map_manager.cpp에 보니 cv::Mat 까지  연결된 i/f를 못 찾겠네요. 요청하신 내용이 그것이지요?
vps에서 뽑아 올릴 수 있는데, apply()같은 interface를 새로 만들어야 하기때문에, 그냥 map_manager.cpp에서 제공 되는것이 합리 적일 것 같습니다. 임재호 선임께서 의견 주세요.

zeozeo7142, 2020-03-25:
Yes, of course I think what I provide is correct and I will make an API to get streetview image(cv::Mat) by dg::ID.

roricljy, 2020-03-26:
I googled it yesterday and found a solution [here](https://answers.opencv.org/question/91344/load-image-from-url/). It works for me.

```
size_t write_data(char *ptr, size_t size, size_t nmemb, void *userdata)
{
    vector<uchar> *stream = (vector<uchar>*)userdata;
    size_t count = size * nmemb;
    stream->insert(stream->end(), ptr, ptr + count);
    return count;
}

cv::Mat getStreeViewImage(dg::ID id, int timeout=10)
{
    std::string img_url = cv::format("http://129.254.87.96:10000/%zu/f", id);

    std::vector<uchar> stream;
    CURL *curl = curl_easy_init();
    curl_easy_setopt(curl, CURLOPT_URL, img_url.c_str()); //the img url
    curl_easy_setopt(curl, CURLOPT_WRITEFUNCTION, write_data); // pass the writefunction
    curl_easy_setopt(curl, CURLOPT_WRITEDATA, &stream); // pass the stream ptr to the writefunction
    curl_easy_setopt(curl, CURLOPT_TIMEOUT, timeout); // timeout if curl_easy hangs,
    CURLcode res = curl_easy_perform(curl); // start curl
    curl_easy_cleanup(curl); // cleanup

    if(res == CURLE_OK && !stream.empty())
    {
        return cv::imdecode(stream, -1); // 'keep-as-is'        
    }
    else
    {
        return cv::Mat();
    }
}
```

zeozeo7142, 2020-03-27:
Thanks for your reference link. I made an API to get streetview image referring to it. It can be operated both around ETRI and Bongeunsa without any options.

closed 2020-03-27 by zeozeo7142