deepguider/DeepGuider #60: [MapManager] direct access of streetview info.

roricljy, 2020-03-25:
We need an api to get StreetView info by streetview ID (not node id)

roricljy, 2020-03-25:
Currently, VPS module returns matched streetview ID's. Therefore, we need to access streetview corresponding to this ID's.

zeozeo7142, 2020-03-25:
I will make APIs to get StreetView and POI info by ID of themselves.

roricljy, 2020-03-26:
I googled it yesterday and found a solution [here](https://answers.opencv.org/question/91344/load-image-from-url/). It works for me.

```
size_t write_data(char *ptr, size_t size, size_t nmemb, void *userdata)
{
    vector<uchar> *stream = (vector<uchar>*)userdata;
    size_t count = size * nmemb;
    stream->insert(stream->end(), ptr, ptr + count);
    return count;
}

cv::Mat getStreeViewImage(dg::ID id, int timeout=10)
{
    std::string img_url = cv::format("http://129.254.87.96:10000/%zu/f", id);

    std::vector<uchar> stream;
    CURL *curl = curl_easy_init();
    curl_easy_setopt(curl, CURLOPT_URL, img_url.c_str()); //the img url
    curl_easy_setopt(curl, CURLOPT_WRITEFUNCTION, write_data); // pass the writefunction
    curl_easy_setopt(curl, CURLOPT_WRITEDATA, &stream); // pass the stream ptr to the writefunction
    curl_easy_setopt(curl, CURLOPT_TIMEOUT, timeout); // timeout if curl_easy hangs,
    CURLcode res = curl_easy_perform(curl); // start curl
    curl_easy_cleanup(curl); // cleanup

    if(res == CURLE_OK && !stream.empty())
    {
        return cv::imdecode(stream, -1); // 'keep-as-is'        
    }
    else
    {
        return cv::Mat();
    }
}
```

roricljy, 2020-03-26:
I checked the recent commit **[MapManager] Add 'getPOI(poi_id)', 'getStreetView(sv_id)'**.
I think it can be revised more..
- It is not easy to check the occurrence of error since it returns dg::StreetView() when it fails.
- It only searches ID within the pre-downloaded local DB. In the user's point of view, however, it will be more convenient if we can access the data without need of explicit pre-download.

zeozeo7142, 2020-03-30:
- I plan to modify all the APIs that the user uses so that errors can be checked with the return value.

- I think it is correct to download the streetview info in advance and search it within the scope. This is because, in order to be able to access the data without need of explicit pre-download, a pre-download is required for the whole country.

roricljy, 2020-04-28:
done!

closed 2020-04-28 by roricljy