deepguider/DeepGuider #62: [???] detection of out of path and localization failure

roricljy, 2020-03-26:
We need to detect if user( or robot) is out of path. If it is the case and localization is correct, we need to regenerate path. However, it may also happen in case when the user is on the right path but the localization is incorrect or fails.

This is one of the key issues of our project and have to be resolved continuously. However, at the moment, I ask each module to implement the following API's.
[Localizer]
 - detection of out of path of **estimated pose** (it may handled in Guidance.. but Localizer also has to care about path. So I ask it to Localizer)
 - detection and announce of localization failure (failure of localization itself)
[Guidance]
 - detection of out of path of **user** (final decision)

Please, leave any comments or suggestions here

seohyunatwork, 2020-03-26:
[Localizer]
2nd bullet: I think this is the case when Localizer gives LOW pose_confidence. This case is when Guidance status is 'GUIDE_LOST' and Exploration should be activated.
[Guidance]
1st bullet: This is same to above. Guidance will track the path and compare the localization info with the path. When Localizer gives 'out of path' node info with HIGH pose_confidence for a while, Guidance will regenerate the path based on the current pose to the destination pose.
2nd bullet: In general case, I think regenerating the path will be easier as the car navigation system than regenerating the path. Going back to the original path is the case when Localizer gives LOW pose_confidence, so 'Exploration' should find out the original path on their own. @ececyh

sinkyv, 2020-03-27:
Let's consider follow steps:
   1. Does a localization result have sufficient confidence?
   2. If not, then we decide the situation as 'GUIDE_LOST' and 'GUIDE_RECOVERY' should be
       activated.
   3. If it does, we should check if a robot is on our path?
   4. If not, then we should select POI recovery or new path generation with the distance
       between current location and  our path.
         4-1. If a robot seems to be able to return to the previous path, POI recovery is better.
         4-2. If location of robot is quite different from our path, new path generation is better. 
   5. If a localization result is good and a robot is on our path but there is no space to go on 
       for a robot because of some permanent obstacles, then new path generation is needed.
   6. If else, the situation is 'GUIDE_NORMAL' and we decide to carry on.

sinkyv, 2020-03-27:
With the upper comment, there are two issues:
   1. How can we decide quickly whether a robot is on our path or not?
   2. How can we recognize a robot is blocked with a permanent obstacle?

closed 2021-09-30 by roricljy