deepguider/DeepGuider #66: [VPS] initial run & setup problem

roricljy, 2020-03-30:
I don't know why but It seems like that we need to execute _deepguider/src/vps/0setup.sh_ again after the computer reboots. I guess this problem is related to make symbolic link to RAM drive.

ccsmm78, 2020-03-30:
Yes, you have pinpointed the cause for it. Because of the ramdisk mount, you need to setup once for each reboot. However, assuming you are using a development-only PC, you can also put the auto mount information in / etc / fstab with the following command. Also, I put this command in 0setup.sh. Therefore, once 0setup.sh is executed for the first time, the mount is maintained even after rebooting.

15 **## Add auto mount information for valid after rebooting your PC**
16 cat /etc/fstab | grep "/media/ramdisk_dg" || echo "none /media/ramdisk_dg tmpfs defaults,size=512M 0 0" | sudo tee -a /etc/fstab
 

**Note that even if only one of the following three commands is executed, 0setup.sh is automatically executed.** 
You can choose a comfortable one :
  - deepguider/example/build_all.sh
  - deepguider/example/vps_test/build_vps.sh
  - deepguider/src/vps/0setup.sh

ccsmm78, 2020-04-01:
Dear Dr. Lee.
Please review the changed code and close this issue.

roricljy, 2020-04-01:
OK, I rebooted the system and run it and it works well.

closed 2020-04-01 by roricljy