deepguider/DeepGuider #67: [VPS] return outputs

roricljy, 2020-03-30:
It looks like that VPS still returns N StreetViews with zero id when there is no nearby streetviews.
[image]

ccsmm78, 2020-03-30:
Yes, that's what I intended. When there is no 'streetview images' to be compared, the result (ID, confidence) will be output as zero. Please recommend if there is another way.

If there is no pointer problem when returning the result from apply (), I can return only the valid result to the list. Even in the case above, null (or '[]') will be returned.

roricljy, 2020-03-30:
That's no problem if it is intended.

closed 2020-03-30 by roricljy