deepguider/DeepGuider #74: [dg_test] code sharing between dg_test and dg_test_ros

roricljy, 2020-04-03:
Currently, similar codes are duplicated at dg_test and dg_test_ros, making it hard to maintain them. It will be more efficient to share same code file.

ccsmm78, 2020-04-03:
catkin workspace와 user의 git clone 위치가 다르기 때문에, 수작업으로 수정해 주어야 할 부분이 있는데, 
[dg_test_ros]에서 setup과 build를 다음과 같이 자동화 하였습니다.
검증 방법은 다음과 같습니다.

git clone 후,
  **1. example에서 build_all.sh 을 실행 :**
        - /home/[user]/catkin_ws_deepguider 라는 catkin workspace가 생성되고, 이곳에 dg_test_ros가 build되어 있습니다.
  2. 혹은, **examples/dg_test_ros 에서 ./setup_rosrun.sh ; build_ros.sh 을 실행 :**
        - /home/[user]/catkin_ws_deepguider 라는 catkin workspace가 생성되고, 이곳에 dg_test_ros가 build되어 있습니다.

이제, ~/catkin_ws_deepguider로 이동 후, roscore, venv 만 실행 후, /dg_run.sh 하면 작동하는것을 볼 수 있습니다(devel/setup.bash 자동 실행됨).

위의 결과를 위해 반영된 내용은 아래와 같습니다. 혹시 불편하시다면, 2회 이전 버전으로 복구하면됩니다. 
       
[dg_test_ros] The environment setup required for rosrun builds was made convenient.

   * New features of examples/dg_test_ros/setup_rosrun.sh :
        - Automatically reflect user's project path in CMakeLists.txt (from .CMakeLists.txt)
        - Automatically create user's hard-coded catkin workspace : ~/catkin_ws_deepguier
        - Automatically reflect user's project path in examples/dg_test_ros/src/dg_test.cpp (from .dg_test.cpp)
        - Automaticaaly insert devel/setup.bash in dg_run.sh (from .dg_run.sh)

   * Add build_ros.sh : catkin_make at user's hard-coded catkin workspace : ~/catkin_ws_deepguier

roricljy, 2020-04-28:
It's done

closed 2020-04-28 by roricljy